robotology-legacy/mex-wholebodymodel #16: [MacOS] timeTest.m seem are slower compiling with make VS Xcode

iron76, 2015-02-13:
When launching timeTest.m, results seems to be slower that what they were in the past. At present results on macOs everything compiled using `make` (default compile options) are:

```
Starting optimised mode trial 
-------------------------- 
mexWholeBodyModel started with robot : icub, Num of Joints : 25 
Robot name icub and World Reference Frame set about l_sole 
Initialisation time : 9.065052e-02 secs 
 Starting Trial...
 Optimised-Mode Trial Total Time : 11.998686 secs
Optimised-Mode Trial Average Time : 1.199869e-03 secs 
```

When compiling with Xcode, again with default options:

```
Starting optimised mode trial 
-------------------------- 
mexWholeBodyModel started with robot : icub, Num of Joints : 25 
Robot name icub and World Reference Frame set about l_sole 
Initialisation time : 8.941473e-02 secs 
 Starting Trial...
 Optimised-Mode Trial Total Time : 2.509663 secs
Optimised-Mode Trial Average Time : 2.509663e-04 secs 
```

naveenoid, 2015-09-18:
Can this be checked by a mac user? @iron76 @francesco-romano @traversaro

traversaro, 2015-09-24:
My iMac machine at IIT is available for testing, it has a working Matlab R2012b installation and you should be able to reproduce this issue (if it is still present). I can send credentials by private mail.

traversaro, 2020-09-29:
We never investigated more in depth this issue.

closed 2020-09-29 by traversaro