robotology-legacy/mex-wholebodymodel #21: Matrix ordering

francesco-romano, 2015-05-13:
Matlab serialises matrices in Column Major ordering while wbi uses the Row major ordering.

Functions must take this issue in consideration (currently I think only the Jacobian computation is affected).

DanielePucci, 2015-05-25:
Hi guys, did we fix this?

DanielePucci, 2015-05-26:
I verify after some debugging: we did not.

naveenoid, 2015-05-26:
Sorry missed this issue..fixing it by tonight.

---
Naveen Kuppuswamy, PhD
Post-doctoral Fellow,
Cognitive Humanoids Lab,
Department of Robotics, Brain and Cognitive Sciences (RBCS),
Istituto Italiano di Tecnologia, Genova, Italy

naveenoid, 2015-05-27:
Would a temporary workaround in the wrapper wbm_jacobian  using reshape()' be sufficient for now? I can push a C++ solution properly.

naveenoid, 2015-05-28:
8d884f32b84bac3d01b245397bdbd9cef55e874b

closed 2015-05-28 by naveenoid