robotology-legacy/mex-wholebodymodel #26: Mex-wholebodymodel functions not explicit with respect of the floating base

gabrielenava, 2015-09-17:
I need to compute numerically the partial derivatives both the mass matrix and the jacobian with respect of a given steady state position. 
Seems  that using wbm functions is possible to compute a variation in mass matrix and jacobian with respect of each joints variable but not with respect of the position and rotation of the floating base.

naveenoid, 2015-09-17:
it looks like ideally you would prefer an implementation of mass-matrix
which takes as an input the complete state (floating base + joint
position).  Do you think if I implement something like
'massMatrixFullState' it would be useful for you?

---
Naveen Kuppuswamy, PhD
Post-doctoral Fellow,
Cognitive Humanoids Lab,
Department of Robotics, Brain and Cognitive Sciences (RBCS),
Istituto Italiano di Tecnologia, Genova, Italy

DanielePucci, 2015-09-17:
@gabrielenava

gabrielenava, 2015-09-17:
yes, i think it will be useful, and i need something similar for the wbm_jacobian.

I have one more question: if i set the robot in a given initial position, can i compute the mass

matrix with respect of the given position and then just change by a delta one of the input in wbm_massmatrix to correctly obtain the new values?

i mean, without changing the entire initial position, make the wbm_updatestate, obtain the new mass matrix, reset the original position, do again update state (which is what i'm doing now). this will save a lot of passages, if it is correct. thanks

Gabriele

A: robotology-playground/mex-wholebodymodel
Cc: Gabriele Nava
Oggetto: Re: [mex-wholebodymodel] Mex-wholebodymodel functions not explicit with respect of the floating base (#26)

naveenoid, 2015-09-18:
I will close this issue as I will address it directly in https://github.com/robotology-playground/mex-wholebodymodel/issues/27

closed 2015-09-18 by naveenoid