robotology-legacy/mex-wholebodymodel #34: Unused initial g in both test scripts

Ganimed, 2015-10-09:
Hi all,

in both testing scripts "forwardDynamicsConsistencyTest.m" and "integrateForwardDynamics.m" there is an initial gravity vector created for the control torques with,

   gInit = wbm_generalisedBiasForces(R_b,p_b,params.qjInit,zeros(25,1),zeros(6,1));

but will never used. For the time-dependent parameter tau it will be used the zeros-function instead of gInit. Is zeros in this case just an initial dummy?
I've found this also in the forwardDynamics function at line 54. Internally g will be calculated but afterwards it will never used or returned.
Can somebody explain me what was the intention, the thought, to calculate there at these places "g" and "gInit"?
Thanks.

traversaro, 2015-10-09:
I guess @naveenoid is the author of the script, so he is the right person to reply!

Ganimed, 2015-10-09:
Ok, I will do that. Thanks.

traversaro, 2015-10-09:
Feel free to update this issue, so everybody can get this info!

Ganimed, 2015-10-09:
I will do that, as when I've got the information.

By the way, in the folder matlab-src there is the folder "worker_functions" where inside are some nice functions for general purposes. In my opinion it is more meaningful if the folder "worker_functions" is renamed into "toolbox". Then everybody understand what is inside.

traversaro, 2015-10-09:
"toolbox" is a bit overloaded in matlab contexts. I would personally prefer something like "utils".

Ganimed, 2015-10-09:
I understand. "utils" is also a good option. We will see ...

Ganimed, 2015-10-09:
"utilities" or just "tools" is also an option ...

naveenoid, 2015-10-22:
Is this still a problem? I thought it was discussed in detail in person? The gInit was used in one of the simulations to set the torque compensating gravity. Im closing this issue for now since we have an overload of others.

closed 2015-10-22 by naveenoid

Ganimed, 2015-10-22:
It was still confusing me, but now I understand. So you used it just temporarily for another simulation. Thanks for the info.

traversaro, 2015-10-22:
For the kinematic testing cleanup that I am doing for solving https://github.com/robotology/mex-wholebodymodel/issues/30 , I remove the unused vector `gInit`.