robotology-legacy/mex-wholebodymodel #37: Warning when running the balancing simulator on Ubuntu 14.04 Matlab 2015b

traversaro, 2015-10-21:
~~~
Warning: The DrawMode property will be removed in a future release. Use the SortMethod property instead. 
> In visualizer_demo (line 39)
  In integrateForwardDynamics (line 132) 
Warning: The DrawMode property will be removed in a future release. Use the SortMethod property instead. 
> In visualizer_demo (line 39)
  In integrateForwardDynamics (line 132) 
Warning: The DrawMode property will be removed in a future release. Use the SortMethod property instead. 
> In visualizer_demo (line 39)
  In integrateForwardDynamics (line 132) 
Warning: The DrawMode property will be removed in a future release. Use the SortMethod property instead. 
> In visualizer_demo (line 39)
  In integrateForwardDynamics (line 132) 
Warning: The EraseMode property is no longer supported and will error in a future release. Use the ANIMATEDLINE function for animating lines and points instead of
EraseMode 'none'. Removing instances of EraseMode set to 'normal', 'xor', and 'background' has minimal impact. 
> In visualizeForwardDynamics (line 191)
  In visualizer_demo (line 55)
  In integrateForwardDynamics (line 132) 
Warning: The EraseMode property is no longer supported and will error in a future release. Use the ANIMATEDLINE function for animating lines and points instead of
EraseMode 'none'. Removing instances of EraseMode set to 'normal', 'xor', and 'background' has minimal impact. 
> In visualizeForwardDynamics (line 191)
  In visualizer_demo (line 55)
  In integrateForwardDynamics (line 132) 
Warning: The EraseMode property is no longer supported and will error in a future release. Use the ANIMATEDLINE function for animating lines and points instead of
EraseMode 'none'. Removing instances of EraseMode set to 'normal', 'xor', and 'background' has minimal impact. 
> In visualizeForwardDynamics (line 191)
  In visualizer_demo (line 55)
  In integrateForwardDynamics (line 132) 
Warning: The EraseMode property is no longer supported and will error in a future release. Use the ANIMATEDLINE function for animating lines and points instead of
EraseMode 'none'. Removing instances of EraseMode set to 'normal', 'xor', and 'background' has minimal impact. 
> In visualizeForwardDynamics (line 191)
  In visualizer_demo (line 55)
  In integrateForwardDynamics (line 132) 
Warning: The EraseMode property is no longer supported and will error in a future release. Use the ANIMATEDLINE function for animating lines and points instead of
EraseMode 'none'. Removing instances of EraseMode set to 'normal', 'xor', and 'background' has minimal impact. 
> In visualizeForwardDynamics (line 191)
  In visualizer_demo (line 55)
  In integrateForwardDynamics (line 132) 
Warning: The EraseMode property is no longer supported and will error in a future release. Use the ANIMATEDLINE function for animating lines and points instead of
EraseMode 'none'. Removing instances of EraseMode set to 'normal', 'xor', and 'background' has minimal impact. 
> In visualizeForwardDynamics (line 191)
  In visualizer_demo (line 55)
  In integrateForwardDynamics (line 132) 
Warning: The EraseMode property is no longer supported and will error in a future release. Use the ANIMATEDLINE function for animating lines and points instead of
EraseMode 'none'. Removing instances of EraseMode set to 'normal', 'xor', and 'background' has minimal impact. 
> In visualizeForwardDynamics (line 191)
  In visualizer_demo (line 55)
  In integrateForwardDynamics (line 132) 
Warning: The EraseMode property is no longer supported and will error in a future release. Use the ANIMATEDLINE function for animating lines and points instead of
EraseMode 'none'. Removing instances of EraseMode set to 'normal', 'xor', and 'background' has minimal impact. 
> In visualizeForwardDynamics (line 191)
  In visualizer_demo (line 55)
  In integrateForwardDynamics (line 132) 
Warning: The EraseMode property is no longer supported and will error in a future release. Use the ANIMATEDLINE function for animating lines and points instead of
EraseMode 'none'. Removing instances of EraseMode set to 'normal', 'xor', and 'background' has minimal impact. 
> In visualizeForwardDynamics (line 191)
  In visualizer_demo (line 55)
  In integrateForwardDynamics (line 132) 
Warning: The EraseMode property is no longer supported and will error in a future release. Use the ANIMATEDLINE function for animating lines and points instead of
EraseMode 'none'. Removing instances of EraseMode set to 'normal', 'xor', and 'background' has minimal impact. 
> In visualizeForwardDynamics (line 191)
  In visualizer_demo (line 55)
  In integrateForwardDynamics (line 132) 
Warning: The EraseMode property is no longer supported and will error in a future release. Use the ANIMATEDLINE function for animating lines and points instead of
EraseMode 'none'. Removing instances of EraseMode set to 'normal', 'xor', and 'background' has minimal impact. 
> In visualizeForwardDynamics (line 191)
  In visualizer_demo (line 55)
  In integrateForwardDynamics (line 132) 
Warning: The EraseMode property is no longer supported and will error in a future release. Use the ANIMATEDLINE function for animating lines and points instead of
EraseMode 'none'. Removing instances of EraseMode set to 'normal', 'xor', and 'background' has minimal impact. 
> In visualizeForwardDynamics (line 191)
  In visualizer_demo (line 55)
  In integrateForwardDynamics (line 132) 
Warning: The EraseMode property is no longer supported and will error in a future release. Use the ANIMATEDLINE function for animating lines and points instead of
EraseMode 'none'. Removing instances of EraseMode set to 'normal', 'xor', and 'background' has minimal impact. 
> In visualizeForwardDynamics (line 191)
  In visualizer_demo (line 55)
  In integrateForwardDynamics (line 132) 
~~~

gabrielenava, 2017-01-24:
Maybe I am wrong, but I remember these warnings have been fixed in the new visualizer. I am not sure since I cannot test it (my matlab version is 2016b). I will leave this open until a proper test

traversaro, 2017-01-24:
Any deprecation warning would be either a warning or an error in a newer version of Matlab, hence the issue was probably solved.

closed 2017-01-24 by traversaro