robotology-legacy/mex-wholebodymodel #39: If `params.leftArmInit` is modified, the initial configuration of the right arm is modified instead

traversaro, 2015-10-21:
If `params.leftArmInit` is set (for example) to :  
~~~
params.leftArmInit  = [ -15   15  0.0  15.9   0.0]';          
~~~
Then in the visualizer the **right** arm is modified instead.

In particular, the initial configuration of the right arm appears to me modified in all the visualization, except for the one on the top right (that instead seems to be correct). 

[image]

traversaro, 2015-10-21:
I reproduce this issue in the `master` branch.

gabrielenava, 2015-10-21:
There's the same problem for the robot balancing on one foot, that should be the left foot but in the visualizer is the right one. I reported this problem in the README file and I wrote that an issue will be opened soon, but then I forgot to open it until now, sorry.

traversaro, 2015-10-21:
No problem @gabrielenava ! I will investigate the issue then.

gabrielenava, 2016-03-10:
fixed on branch feature/torques_linear, at #51e80d41e407938a6edc2ab86ded8cf57c57ddb2

closed 2016-03-10 by gabrielenava