robotology-legacy/mex-wholebodymodel #40: Simplify repository structure

traversaro, 2015-10-21:
Hi everyone, one of the source of confusion related to this repository is that it contains mex/functions/scripts fulfilling two different purposes: 
 * C++ code and matlab code in `matlab-src/whole_body_model_functions` : Provide a matlab interface to the `yarpWholeBodyModel` C++ class (original purpose of the repo, `mex-wholebodymodel` properly said)
 * All the other matlab code: implement simulation and control code in matlab, using the above mentioned interface

To clarify this I propose to move all the code in two directories : 
 * `mex-wholebodymodel` (like the repo itself) for all the interface code.
 * `simulation` for all the simulation & control related code.  

With respect to https://github.com/robotology/mex-wholebodymodel/issues/20 , I think it then make sense to properly install all the interface-related mex / code . 
It could make sense instead to avoid installation of simulation code, so that it can be directly edited and used from the repo. 

It could make also sense to move all the functions in proper MATLAB packages, but I guess it would be a big step and I would then prefer to defer it.

traversaro, 2015-10-21:
Feedback request: @francesco-romano @DanielePucci @gabrielenava @naveenoid @Ganimed

traversaro, 2015-10-21:
@Ganimed  I think it would make sense for your class to be in the `mex-wholebodymodel` directory.

naveenoid, 2015-10-22:
Agreed very strongly. I am not at all happy to mix matlab and C++ directly and talked about this in the original design a year ago.

DanielePucci, 2015-10-22:
I would separate the controllers anyway. Since we already have a WBI-Toolboxcontrollers repository, why don't we put the controllers there?

traversaro, 2015-10-22:
It make sense to separate also controllers, but I consider it to be an issue orthogonal to this one.

DanielePucci, 2015-10-22:
Ok, it seemed to me that when you wrote
~~~
simulation for all the simulation & control related code.
~~~
you were referring to the controllers also.

traversaro, 2015-10-22:
Yes, until they are properly moved.

francesco-romano, 2015-11-04:
A first attempt has been done in https://github.com/robotology/mex-wholebodymodel/pull/53

The only thing left is to properly change name to the directories (if it make sense).
Otherwise @traversaro if you think it is enough for now we can close this issue.

traversaro, 2015-11-04:
Ok for me, cleaning the `matlab-src` directory is an important but separate job.

closed 2015-11-04 by traversaro

Ganimed, 2015-11-04:
**@traversaro:** Yes I will do that, when the class is finished. The last two weeks before I was in Austria and in this meantime the amount of discussions was exploding here. So I lost the oversight what is going on. But I will get a briefing from **naveenoid** when he has time. I will move the WBM-Class to the _mex-wholebodymodel_ directory when it is finished. Else it does not make sense. Since yesterday the initialization methods has been changed, I have to make a changes to the constructor of the class.

I have seen that the _mex-wholebodymodel_ directory is well structured. I like the utilities folder! :) (Why not so from begin on? But I do not know the complete history of the development process.) A few weeks ago I made also an _+utilities_ folder in my class directory which Matlab will recognize it as a sub-namespace. This is maybe better than a normal folder, to prevent loading same function-names of different repositories (e.g. skew). But this are details and can be changed.

What I have seen in last weeks here, especially when I was in Vienna, it looked from outside that the discussions were bit chaotic. What we need, I guess, is a "project management". Let us sit together and talk about how are the future plans (methods, structures, changes, etc.) for the mex-WBM. Lets draw it down on the blackboard (e.g. folder structure, folder names, position of the folders, names of the methods, etc.). Mind mapping would be helpful for that which we can store, share and extend our ideas. I'm using for that Freemind.