robotology-legacy/mex-wholebodymodel #41: Don't keep `.mat` files in the repo

traversaro, 2015-10-21:
I think it make sense to remove all the `.mat` files from the repo. 

 `.mat` files are binaries for which is quite difficult to track the modification, for example while checking the repo history. 

I think all the `.mat` files commited can be replaced by `.m` files produced by [Simulink.saveVars](http://it.mathworks.com/help/simulink/slref/simulink.savevars.html) function. We will lose some precision in the stored results, but the overall benefit seems clear to me.

naveenoid, 2015-10-22:
The only reason for this was the old issue regarding jointLimits. If that is fixed. it can be removed.

traversaro, 2015-10-27:
For the same reason we don't need to have `.fig` files in the repo. @gabrielenava

gabrielenava, 2015-10-28:
ok, capito

gabrielenava, 2016-03-10:
all the .mat and .fig files have been eliminated and are not necessary anymore, so I think we can close this issue

closed 2016-03-10 by gabrielenava