robotology-legacy/mex-wholebodymodel #42: [Simulation] Don't Repeat Yourself : Avoid copy pasting code, use functions

traversaro, 2015-10-21:
Some logic that is currently copy-pasted across a lot of functions/scripts, and need to be consolidated in functions: 

## Constrained Dynamics with Torque as a input 

* [ ] Tick when fixed 

This function is already available in the `forwardDynamics` function in https://github.com/robotology/mex-wholebodymodel/blob/master/matlab-src/forwardDynamics.m , but for some reason is not used in the balancing/controller simulation.  Instead a `forwardDynamics` function with the same signature but with different semantics ( O_o ) is implemented in https://github.com/robotology/mex-wholebodymodel/blob/master/matlab-src/mex-wholebodymodel_balancing/forwardDynamics.m . We definitively need to consolidate this.  

## Quaternion to Rotation conversion

* [ ] Tick when fixed 

Duplicated in at least 6 files : 
https://github.com/robotology/mex-wholebodymodel/search?utf8=%E2%9C%93&q=2*qt_b_mod_s*skew%28qt_b_mod_r%29&type=Code

## Simulation to mex-wholebodymodel (and reverse) conversion 

* [ ] Tick when fixed 

The position of the robot is a (7+n_J) vector for the simulation code (
containing the quaternion of the base orientation) and a triple (RotationMatrix, BaseOriginPosition, 
JointPos ) (where RotationMatrix is a 3x3 matrix, BaseOriginPosition a 3 vector and JointPos a n_J vector). We hardcode how this two representation are mapped, we need to write two functions to encapsulate this mapping. 

Duplicated in at least 8 files :
https://github.com/robotology/mex-wholebodymodel/search?utf8=%E2%9C%93&q=x_b++%3D+chi%281%3A3%2C%3A%29%3B++qt_b+%3D+chi%284%3A7%2C%3A%29%3B

traversaro, 2015-10-23:
Regarding the quaternion to rotation, two formulas are used in the code:
~~~
dcm = eye(3) + 2*qt_b_mod_s*skew(qt_b_mod_r) + 2 * skew(qt_b_mod_r)^2;
~~~
in the [quite used `frame2pos` function](https://github.com/robotology/mex-wholebodymodel/search?utf8=%E2%9C%93&q=frame2posrot&type=Code) and 
~~~
dcm = eye(3) - 2*qt_b_mod_s*skew(qt_b_mod_r) + 2 * skew(qt_b_mod_r)^2;
~~~
In the rest of the code. As far as I remember the one with the minus is the correct one, but I think it is good if we all agree on which fix is the one most correct. In strict sense, both are valid (in the sense that they form a consistent mapping between the quaternions and SO(3) .

DanielePucci, 2015-10-23:
Guys,  I'm really frustrated,  since I remember that I personally worked on that, and II tried to fix this stuff with Gabriele.  I don't know why these obscure formulae keep going around. I remember that the first one should be right, instead [1, p. 101, Eq. 3.8]. Anyways, please try to write these functions once for all in appropriate separate files,  so we can debug them only once. 

[1] THÈSE Contributions au contrôle automatique de véhicules aériens - Sophia Antipolis

traversaro, 2015-10-23:
I checked and apparently the formula with a `plus` is the one consistent with Rodriguez Formula.

traversaro, 2015-10-23:
I also checked and apparently `mex-wholebodymodel_balancing` code is not using the `minus` version anywhere.

DanielePucci, 2015-10-23:
Ack.

naveenoid, 2015-10-23:
I have no idea why the incorrect version crept in. I remember fixing this a while ago also with @gabrielenava.  I guess with automated tests this will stop being a problem.

traversaro, 2015-10-23:
Given that it was the source of the error in https://github.com/robotology/mex-wholebodymodel/issues/30 and an important piece of computation in general, I will now investigate how to consolidated all the different functions implementing the "Constrained Dynamics with Torque as a input".

traversaro, 2016-03-24:
@gabrielenava  Can you fix the last point in https://github.com/robotology/mex-wholebodymodel/issues/42#issue-112623327 ?

commit 1ea9bab1 (2016-03-29): fixed #42
closed 2016-03-29 by gabrielenava