robotology-legacy/mex-wholebodymodel #43: [Interface] Remove wbm_setWorldLink and wbm_setWorldFrame ?

traversaro, 2015-10-21:
A lot of user have been confused by the semantics of this two functions https://github.com/robotology/mex-wholebodymodel/issues/26 and now it is possible to use all the functions passing all the state https://github.com/robotology/mex-wholebodymodel/issues/27 . 

If no code it is supposed to use this function, perhaps it is a better idea to just remove them from the interface, to avoid confusion/misuse by the users? 

After that, for symmetry with all existing calls, it could make sense to have an updateState in which you pass all the state, with also the base position ?

traversaro, 2015-10-21:
cc @naveenoid @gabrielenava  opinions?

traversaro, 2015-10-22:
This would also have the side benefits of removing the hardcoded "l_sole" in the code ( https://github.com/robotology/mex-wholebodymodel/blob/6603ab444e828f05bcb71c0317e4421e97736ac8/src/modelstate.cpp#L41 ),

traversaro, 2015-10-23:
Another example of confusion regarding `wbm_setWorldLink` semantics : https://github.com/robotology/mex-wholebodymodel/issues/47#issuecomment-150511583 .

DanielePucci, 2015-10-23:
I completely agree with you: passing the whole state is the right way to go. How do we pass this state for  SE(3)" Do we pass quaternions or what? @naveenoid  opinions?

traversaro, 2015-10-23:
We are currently passing rotation matrices.

DanielePucci, 2015-10-23:
Now we are talking [1]

[1]: http://www.youtube.com/watch?v=k17KH6F63aE&t=1m24s

naveenoid, 2015-10-25:
I will most likely remove this once PR https://github.com/robotology/mex-wholebodymodel/pull/49 is merged

naveenoid, 2015-10-28:
Tackled in ttps://github.com/robotology/mex-wholebodymodel/pull/52

closed 2015-11-03 by traversaro