robotology-legacy/mex-wholebodymodel #46: [interface] ComponentManager is allocated and never deleted

traversaro, 2015-10-22:
The instance of the `ComponentManager` is allocated in https://github.com/robotology/mex-wholebodymodel/blob/77139ce83e22b766085d83542c7a1caedde74cdf/src/componentmanager.cpp#L40 , but is never properly deleted.

traversaro, 2015-10-22:
Fixed this bug (and all other problems with memory allocation and deallocation, that prevented the mex to be unloaded from matlab in a clean way) in three commits: 
* https://github.com/robotology/mex-wholebodymodel/commit/f60c9ac833a0ea80b0731ac2386af7d598ce96f5 
* https://github.com/robotology/mex-wholebodymodel/commit/0437e086ba93754111f219b1a86aae8113722958
* https://github.com/robotology/mex-wholebodymodel/commit/5f68f764ba4855aa906d68154a683ebfc8cf35c8 

I will merge it together with the fix to the kinematic energy conservation tests, then we can close this issue.

naveenoid, 2015-10-25:
good catch! Not sure when it crept in. :+1:

closed 2015-10-26 by traversaro