robotology-legacy/mex-wholebodymodel #47: [interface] The base part of wbm_getState is broken

traversaro, 2015-10-23:
In particular, the base state was undefined because it was taken from the returned value of a function that was not returning anything (see https://github.com/robotology/mex-wholebodymodel/commit/0437e086ba93754111f219b1a86aae8113722958#commitcomment-13937981 ).
I guess the function is always returning the identity transform in practice. 
Fixed in https://github.com/robotology/mex-wholebodymodel/commit/0437e086ba93754111f219b1a86aae8113722958 , still need to be fixed in master.

traversaro, 2015-10-23:
cc @naveenoid @gabrielenava

traversaro, 2015-10-23:
To make sure that everything is properly working, I suggest that we do some consistency tests together with `wbm_setWorldFrame``wbm_updateState` and `wbm_forwardKinematics`  .

traversaro, 2015-10-23:
Anyway the effect of the simulations is limited, as getState was used only to get the initial configuration of the base.

gabrielenava, 2015-10-23:
I don't think I have really understand this problem. I mean, in the balancing controller wbm_getState is used only at the beginning and only to obtain the identity since wbm_setWorldLink sets exactly the identity transform. But if I put wbm_getState in forwardDynamics just to see what happens the quaternion returned by getState  changes, it's still a valid quaternion and the associated rotation matrix is not the identity so I don't think I understand this problem properly

traversaro, 2015-10-23:
I'll explain what I found in the code: 
[`wbm_getState`](https://github.com/robotology/mex-wholebodymodel/blob/6603ab444e828f05bcb71c0317e4421e97736ac8/matlab-src/whole_body_model_functions/wbm_getState.m#L19) calls (with [some magic](https://github.com/robotology/mex-wholebodymodel/blob/77139ce83e22b766085d83542c7a1caedde74cdf/src/componentmanager.cpp#L61) )
[ModelGetState::computeFast](https://github.com/robotology/mex-wholebodymodel/blob/77139ce83e22b766085d83542c7a1caedde74cdf/src/modelgetstate.cpp#L199) that itself calls 
[`modelState->getRootWorldRotoTranslation()` to get the rototranslation, but this function is not returning anything!](https://github.com/robotology/mex-wholebodymodel/blob/77139ce83e22b766085d83542c7a1caedde74cdf/src/modelstate.cpp#L350)

The fact that you mention `wbm_setWorldLink` (that should not be used in simulation, see https://github.com/robotology/mex-wholebodymodel/issues/43) make me wonder if perhaps `ModelGetState::compute` is called, and that function seems is not affected by the bug. 

Anyway a general comment: the fact that a computed quantity changes and is not the defuault does not mean that it is correctly computed!

naveenoid, 2015-10-28:
Tackled in https://github.com/robotology/mex-wholebodymodel/pull/52

closed 2015-11-03 by traversaro