robotology-legacy/mex-wholebodymodel #61: WBM-Class is ready for merging into the master

Ganimed, 2015-12-18:
Hi all,

I want to announce that the new WBM-Class branch is ready for merging into the master. This new wrapper-structure can be considered as the next generation for the mex-WBM.

__What is new there:__

* OO approach.
* Namespaces instead of paths. This prevents also errors when other methods have by accident the same name.
* Rewritten and optimized the utilities methods
* The initialization of the model is now clear defined and easier to use than before
* Error-checks (dimensions, size, etc.) at the most important positions in the class
* Speed optimized for run-time (thanks to the book: Accelerating MATLAB Performance: 1001 tips to speed up MATLAB programs  [I love this book! ] )
* Removed all Matlab-warnings while plotting the results (since R2014b, Matlab has changed a lot in the background for the graphics-GUI, i.e. using OpenGL, and replaced several functions and variables).
* Introduced a fastForwardDynamics function which is decoupled from the class to get more speed for time-critical computations.

__What is still missing:__

* visualizeForwadDynamics function.
* Full documentation in M2HTML.

(Both will be done after Christmas.)

_To check:_

* Since "chi" can getting very large in the end, use a GPU-matrix/array for chi instead of a normal. This can maybe increase the performance for further computations afterwards.

__One open bug:__

There is still one bug during the initialization process, which I'm not able to remove in Linux. This error happens also in the Matlab-files of Naveen. Im using Linux Mint 17.3 Rosa and I tested it under Matlab R2014b and R2015b. I get the same error in both versions. I don't know if the error is also under MaxOS and MS-Windows. We have to check it. The error happens when ```wbm_modelInitialise();``` is called in the optimized mode. If no name is set then the function should then read the default robot name from environment variable ```YARP_ROBOT_NAME```. But it doesn't under Linux and get the following error message:

```
...
[ERROR]yarpWholeBodyModel: robot option not found 
WBI unable to initialise (usually means unable to connect to chosen robot)
...
```

I have also checked my configuration and the environment variables under ```~/.profile```. I have everything set what was given in the install instruction.
Do I still miss something?

Here is the excerpt of my configuration in _.profile_:

```
# Environment variables for the CoDyCo-Superbuild project:
export CODYCO_SUPERBUILD_ROOT=/home/ganymed/Library/codyco-superbuild
export PATH=$PATH:$CODYCO_SUPERBUILD_ROOT/build/install/bin
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$CODYCO_SUPERBUILD_ROOT/build/install/lib
export YARP_DATA_DIRS=$YARP_DATA_DIRS:$CODYCO_SUPERBUILD_ROOT/build/install/share/codyco
# environmental variable for the WBI-Toolbox:
export YARP_ROBOT_NAME="icubGazeboSim"
```

In my repository I have made a temporary file under _matlab-src/tests_, called *timeTest_2.m*, to reproduce the error without changing other files. There I have deactivated the normal method ```wbm_modelInitialise('icubGazeboSim');```
and replaced it with ```wbm_modelInitialise();```. Please feel free to test it also under your OS, to see if you are getting the same error as me.
Thanks.

P.S. By the way, if we don't see us, I wish you a merry Christmas and a Happy New Year!

Ganimed, 2015-12-18:
Before I forget, I have two files adapted to the new structure. One as an example and the other for time-benchmarking. They are stored under _+WBM/test_ and _+WBM/examples_. The speed of the classes depends on the CPU of the Computer and from the used Matlab-Version. I optimized what I could. But since Matlab is a script-language, I want to note that the parser needs always a bit more time for interpreting a class-structure then a single function file. So for time-critical computations, it is better to use the methods of Naveen.

To see how the new WBM-package is structured, there is a folder called _UML_. There you will find a PDF-file with an UML diagram.

naveenoid, 2015-12-21:
I think this whole issue needs to  be tracked from here : https://github.com/robotology/mex-wholebodymodel/pull/62

closed 2017-01-30 by gabrielenava