robotology-legacy/mex-wholebodymodel #64: Warning: quaternions numerical integration errors

gabrielenava, 2016-03-09:
In the Matlab torque controllers, quaternions are integrated using the Matlab function ode15s. We observed that there's a numerical integration error in the quaternions calculation, whose magnitude is constant in time and depends on the integration tolerances. This error propagates to the rotation matrix, which is used to calculate jacobians, centroidal momentum, mass matrix and forward kinematics. With a relative error tolerance 'RelTol' = 1e-5 and a vector of absolute error tolerances 'AbsTol' = 1e-5, the magnitude of this error is 1e-6. If it is necessary, one can reduce the error reducing the integration tolerances.

traversaro, 2016-03-09:
After integration, quaternions are never projected back to the unit hyper-sphere (i.e. normalized)?

traversaro, 2016-03-09:
What are the long-term effects of this integration errors?

naveenoid, 2016-03-09:
One option could be to try to normalise quaternions both before and after
computation of the quaternion derivative within your dynamics function.
This might help with the deviation induced during one integrator loop.

gabrielenava, 2016-03-09:
They are not normalized. It seems that there are no significant effects on system's dynamics, I think because the magnitude of this error is not increasing in time.

gabrielenava, 2016-03-09:
Actually, the time derivative of quaternions has a correction term, I think to ensure its validity. Maybe this is the reason why the integration error is not increasing

traversaro, 2016-03-09:
The magnitude of which error is not increasing? The magnitude of the *quaternion error* (i.e. distance from the unit hyper-sphere) or the integration error? 
I assume that you are referring to the *quaternion error*. 
I strongly suggest to normalize the quaternion coming out from the integration, even if the correction term is ensuring that the *quaternion error* is remaining constant. All the quaternion handling function (such as conversion to rotation matrices or euler angles) assume unit quaternions, and estimating the numerical effect of an error of `1e-6` on the quaternion in the output of this functions is not trivial. Normalizing the quaternion should bring the quaternion error to have a value similar to the [machine epsilon](https://en.wikipedia.org/wiki/Machine_epsilon), for our case `~1e-16`.

gabrielenava, 2016-03-10:
I normalize the quaternion after the integration and this fixed the bug. Also,  I managed to fix another old bug #39. I'll push the corrections in my branch for now, I'll fix other bugs then I'll be ready for pull request.

closed 2016-03-10 by gabrielenava