robotology-legacy/mex-wholebodymodel #66: error while use wbm_jacobian

VModugno, 2016-04-27:
Hi guys,

i had some troubles using wbm_jacobian. When i try to compute the jacobian of a frame specified in the urdf of icubGazeboSim but different from 'com' , 'l_sole' , 'r_sole' i receive an error massage saying that the frame is not listed. how can i fix this? 
thank you for your help

@serena-ivaldi  @UChervet

traversaro, 2016-04-28:
Hi, can you provide the line of code that are generating the error, the exact error that you get and your environment (OS, Matlab version)? Thanks!

gabrielenava, 2017-01-24:
I tried to specify a frame different from the CoM and the feet, and wbm_jacobian worked fine:

```

>> wbm_jacobian('l_upper_leg_contact')

ans =

  Columns 1 through 12

    1.0000   -0.0000    0.0000   -0.0000   -0.1987   -0.0690         0         0         0         0         0         0
   -0.0000    1.0000   -0.0000    0.1987   -0.0000   -0.0683         0         0         0         0         0         0
    0.0000   -0.0000    1.0000    0.0690    0.0683    0.0000         0         0         0         0         0         0
         0         0         0    1.0000   -0.0000    0.0000         0         0         0         0         0         0
         0         0         0   -0.0000    1.0000   -0.0000         0         0         0         0         0         0
         0         0         0    0.0000   -0.0000    1.0000         0         0         0         0         0         0

  Columns 13 through 24

         0         0         0         0         0         0         0    0.0816   -0.0002   -0.0009         0         0
         0         0         0         0         0         0         0    0.0000    0.0700   -0.0596         0         0
         0         0         0         0         0         0         0   -0.0423    0.0009   -0.0002         0         0
         0         0         0         0         0         0         0    0.0000    0.9744   -0.2250         0         0
         0         0         0         0         0         0         0   -1.0000    0.0000   -0.0000         0         0
         0         0         0         0         0         0         0   -0.0000    0.2250    0.9744         0         0

  Columns 25 through 31

         0         0         0         0         0         0         0
         0         0         0         0         0         0         0
         0         0         0         0         0         0         0
         0         0         0         0         0         0         0
         0         0         0         0         0         0         0
         0         0         0         0         0         0         0

```
for this reason, and because we don't have a well documented error, I think we can close the issue.

closed 2017-01-24 by gabrielenava