robotology-legacy/mex-wholebodymodel #67: Cmake: check dependencies

francesco-romano, 2016-05-02:
There are a lot of dependencies required by this project.
I think that a lot of these are not required (neither now, nor when the project was first created).

The CMakeLists.txt should then be cleaned.

gabrielenava, 2016-07-11:
I deleted one not required dependency (paramHelp) that caused a cmake error while compiling codyco-superbuild.

gabrielenava, 2017-01-26:
I tried to rewrite the CMakeLists.txt eliminating the unnecessary code, but I have no idea if dependencies are really necessary or not. this is the updateted [CMakeLists.txt](https://github.com/robotology/mex-wholebodymodel/blob/master/CMakeLists.txt)

gabrielenava, 2017-01-30:
With the help of @Ganimed, I removed all unnecessary dependencies from CMakeLists so we can close this issue (see 45f8118c58497cb2b28b76f111e37b20b081823b and c882bf4ad74778abcfd44ac1d271864977366b68)

closed 2017-01-30 by gabrielenava