robotology-legacy/mex-wholebodymodel #70: Matlab crash when running the ctest

gabrielenava, 2016-05-25:
I tried to run the ctest on the **fixWin** branch. The ctest fails with the following error: 

______________________________________________________________________
Fault Count: 1
1: 
1: 
1: Abnormal termination:
1: Segmentation violation
1: 
1: Register State (from fault):
1:   RAX = c03e929fbc8d0fef  RBX = 00007fe3df7f19b0
1:   RCX = 00007fe3d3643ba0  RDX = 00007fe3d3643ba0
1:   RSP = 00007fe3df7f17c0  RBP = 00007fe3df7f1810
1:   RSI = 00007fe3d2d3d5c8  RDI = 00007fe3df7f19b0
1: 
1:    R8 = 0000000000000000   R9 = 00007fe3d341e210
1:   R10 = 00007fe3d3077e08  R11 = 0000000000000001
1:   R12 = 0000000000000006  R13 = 00007fe3d2d3d5c8
1:   R14 = 0000000000000000  R15 = 00007fe3d3213890
1: 
1:   RIP = 00007fe3f71587f0  EFL = 0000000000010282
1: 
1:    CS = 0033   FS = 0000   GS = 0000
1: 
1: Stack Trace (from fault):
1: [  0] 0x00007fe3f86611de    /usr/local/MATLAB/R2012b/bin/glnxa64/libmwfl.so+00516574 _ZN2fl4diag15stacktrace_base7captureERKNS0_14thread_contextEm+000158
1: [  1] 0x00007fe3f86624b2    /usr/local/MATLAB/R2012b/bin/glnxa64/libmwfl.so+00521394
1: [  2] 0x00007fe3f8663ffe    /usr/local/MATLAB/R2012b/bin/glnxa64/libmwfl.so+00528382 _ZN2fl4diag13terminate_logEPKcRKNS0_14thread_contextE+000174
1: [  3] 0x00007fe3f7950093   /usr/local/MATLAB/R2012b/bin/glnxa64/libmwmcr.so+00557203 _ZN2fl4diag13terminate_logEPKcPK8ucontext+000067
1: [  4] 0x00007fe3f794cb9d   /usr/local/MATLAB/R2012b/bin/glnxa64/libmwmcr.so+00543645
1: [  5] 0x00007fe3f794e835   /usr/local/MATLAB/R2012b/bin/glnxa64/libmwmcr.so+00550965
1: [  6] 0x00007fe3f794ea55   /usr/local/MATLAB/R2012b/bin/glnxa64/libmwmcr.so+00551509
1: [  7] 0x00007fe3f794f0fe   /usr/local/MATLAB/R2012b/bin/glnxa64/libmwmcr.so+00553214
1: [  8] 0x00007fe3f794f295   /usr/local/MATLAB/R2012b/bin/glnxa64/libmwmcr.so+00553621
1: [  9] 0x00007fe3f613c340              /lib/x86_64-linux-gnu/libpthread.so.0+00066368
1: [ 10] 0x00007fe3f71587f0 /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+04515824
1: [ 11] 0x00007fe3f7154a1d /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+04499997
1: [ 12] 0x00007fe3f717eafc /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+04672252
1: [ 13] 0x00007fe3f715934d /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+04518733
1: [ 14] 0x00007fe3f70c7810 /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+03921936
1: [ 15] 0x00007fe3f70cb1a6 /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+03936678
1: [ 16] 0x00007fe3f70cb5dc /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+03937756
1: [ 17] 0x00007fe3f6ee7fec /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+01957868
1: [ 18] 0x00007fe3f6ee40d3 /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+01941715
1: [ 19] 0x00007fe3f6ee4ed7 /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+01945303
1: [ 20] 0x00007fe3f6f50760 /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+02385760
1: [ 21] 0x00007fe3f769ea4b /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_dispatcher.so+00596555 _ZN8Mfh_file11dispatch_fhEiPP11mxArray_tagiS2_+000539
1: [ 22] 0x00007fe3f711f206 /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+04280838
1: [ 23] 0x00007fe3f711f97a /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+04282746
1: [ 24] 0x00007fe3f71204ea /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+04285674
1: [ 25] 0x00007fe3f6f834cd /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+02593997
1: [ 26] 0x00007fe3f6faed22 /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+02772258
1: [ 27] 0x00007fe3f6faee4f /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+02772559
1: [ 28] 0x00007fe3f70cbb30 /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+03939120
1: [ 29] 0x00007fe3f6ee7fec /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+01957868
1: [ 30] 0x00007fe3f6ee40d3 /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+01941715
1: [ 31] 0x00007fe3f6ee4ed7 /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+01945303
1: [ 32] 0x00007fe3f6f50760 /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+02385760
1: [ 33] 0x00007fe3f769ea4b /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_dispatcher.so+00596555 _ZN8Mfh_file11dispatch_fhEiPP11mxArray_tagiS2_+000539
1: [ 34] 0x00007fe3f6f32e56 /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+02264662
1: [ 35] 0x00007fe3f6ebec1d /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+01788957
1: [ 36] 0x00007fe3f6ee724e /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+01954382
1: [ 37] 0x00007fe3f6ee40d3 /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+01941715
1: [ 38] 0x00007fe3f6ee4ed7 /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+01945303
1: [ 39] 0x00007fe3f6f50760 /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+02385760
1: [ 40] 0x00007fe3f769ea4b /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_dispatcher.so+00596555 _ZN8Mfh_file11dispatch_fhEiPP11mxArray_tagiS2_+000539
1: [ 41] 0x00007fe3f6f32e56 /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+02264662
1: [ 42] 0x00007fe3f6ebec1d /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+01788957
1: [ 43] 0x00007fe3f6ee724e /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+01954382
1: [ 44] 0x00007fe3f6ee40d3 /usr/local/MATLAB/R2012b/bin/glnxa64/libmwm_interpreter.so+01941715
[... 36 lines cut ...]
1: 
1: 
1: If this problem is reproducible, please submit a Service Request via:
1:     http://www.mathworks.com/support/contact_us/
1: 
1: A technical support engineer might contact you with further information.
1: 
1: Thank you for your help.** This crash report has been saved to disk as /home/gnava/matlab_crash_dump.14273-1 **
1: 
1: 
1: 
1: MATLAB is exiting because of fatal error
1/1 Test #1: matlab_mex_wholebodymodel_tests ...***Exception: Other163.34 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) = 163.34 sec

The following tests FAILED:
	  1 - matlab_mex_wholebodymodel_tests (OTHER_FAULT)
Errors while running CTest

_________________________________________________________________________

apparently, the error is also in the master branch

Ganimed, 2016-05-26:
Maybe the mex-file is causing somewhere a null-pointer exception. Re-check the source code, i.e. all pointers and their allocations and deallocations. In Linux there exist a great tool for that, called "Valgrind".

Ganimed, 2016-05-26:
Is the error only in Linux/Mac OS X?

gabrielenava, 2017-01-24:
@Ganimed I run the ctest in your WBM branch, and I still had the segmentation violation problem:

```
1: Fault Count: 1
1: 
1: 
1: Abnormal termination:
1: Segmentation violation
1: 
1: Register State (from fault):
1:   RAX = 4006ac5c6555bda3  RBX = 00007ff675279d98
1:   RCX = 00007ff686c57e60  RDX = 00007ff686c57e80
1:   RSP = 00007ff686c57bc0  RBP = 00007ff686c57bd0
1:   RSI = 00007ff675279d98  RDI = 00007ff681651f30
1: 
1:    R8 = 00007ff686c57e40   R9 = 00007ff686c57e70
1:   R10 = 0000000000000000  R11 = 0000000000000001
1:   R12 = 00007ff681651f30  R13 = 00007ff686c57e60
1:   R14 = 00007ff686c57e50  R15 = 00007ff686c57e40
1: 
1:   RIP = 00007ff693ee1940  EFL = 0000000000010202
1: 
1:    CS = 0033   FS = 0000   GS = 0000
1: 
1: Stack Trace (from fault):
1: [  0] 0x00007ff693ee1940 /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+11843904
1: [  1] 0x00007ff693ee1ac5 /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+11844293
1: [  2] 0x00007ff693846a56 /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+04917846
1: [  3] 0x00007ff693843410 /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+04903952
1: [  4] 0x00007ff69383e855 /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+04884565
1: [  5] 0x00007ff69383eb69 /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+04885353
1: [  6] 0x00007ff69384320d /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+04903437
1: [  7] 0x00007ff6938432e2 /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+04903650
1: [  8] 0x00007ff69393a688 /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+05916296
1: [  9] 0x00007ff69393cb2f /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+05925679
1: [ 10] 0x00007ff693dbb10e /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+10637582
1: [ 11] 0x00007ff693eaae33 /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+11619891
1: [ 12] 0x00007ff69707d8a3 /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_dispatcher.so+00768163 _ZN8Mfh_file16dispatch_fh_implEMS_FviPP11mxArray_tagiS2_EiS2_iS2_+00000947
1: [ 13] 0x00007ff69707e16e /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_dispatcher.so+00770414 _ZN8Mfh_file11dispatch_fhEiPP11mxArray_tagiS2_+00000030
1: [ 14] 0x00007ff693eb8847 /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+11675719
1: [ 15] 0x00007ff693eb8aab /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+11676331
1: [ 16] 0x00007ff693f1e461 /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+12092513
1: [ 17] 0x00007ff693844930 /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+04909360
1: [ 18] 0x00007ff693846c3c /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+04918332
1: [ 19] 0x00007ff693843410 /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+04903952
1: [ 20] 0x00007ff69383e855 /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+04884565
1: [ 21] 0x00007ff69383eb69 /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+04885353
1: [ 22] 0x00007ff69384320d /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+04903437
1: [ 23] 0x00007ff6938432e2 /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+04903650
1: [ 24] 0x00007ff69393a688 /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+05916296
1: [ 25] 0x00007ff69393cb2f /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+05925679
1: [ 26] 0x00007ff693dbb10e /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+10637582
1: [ 27] 0x00007ff693eaae33 /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+11619891
1: [ 28] 0x00007ff69707d8a3 /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_dispatcher.so+00768163 _ZN8Mfh_file16dispatch_fh_implEMS_FviPP11mxArray_tagiS2_EiS2_iS2_+00000947
1: [ 29] 0x00007ff69707e16e /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_dispatcher.so+00770414 _ZN8Mfh_file11dispatch_fhEiPP11mxArray_tagiS2_+00000030
1: [ 30] 0x00007ff693eb8847 /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+11675719
1: [ 31] 0x00007ff693eb8aab /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+11676331
1: [ 32] 0x00007ff693f1e461 /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+12092513
1: [ 33] 0x00007ff693844930 /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+04909360
1: [ 34] 0x00007ff693846c3c /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+04918332
1: [ 35] 0x00007ff693843410 /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+04903952
1: [ 36] 0x00007ff69383e855 /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+04884565
1: [ 37] 0x00007ff69383eb69 /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+04885353
1: [ 38] 0x00007ff69384320d /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+04903437
1: [ 39] 0x00007ff6938432e2 /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+04903650
1: [ 40] 0x00007ff69393a688 /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+05916296
1: [ 41] 0x00007ff69393cb2f /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+05925679
1: [ 42] 0x00007ff693dbb10e /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+10637582
1: [ 43] 0x00007ff693eaae33 /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_lxe.so+11619891
1: [ 44] 0x00007ff69707d8a3 /usr/local/MATLAB/R2016b/bin/glnxa64/libmwm_dispatcher.so+00768163 _ZN8Mfh_file16dispatch_fh_implEMS_FviPP11mxArray_tagiS2_EiS2_iS2_+00000947
[... 65 lines cut ...]
actually, unlike the previous time, the test started without problems:

```
test 1
    Start 1: matlab_mex_wholebodymodel_tests

 Test command: /usr/local/MATLAB/R2016b/bin/matlab "-nodesktop" "-nojvm" "-nodisplay" "-r" "runWBMTests"
 Test timeout computed to be: 1500
 
                             < M A T L A B (R) >
                   Copyright 1984-2016 The MathWorks, Inc.
                    R2016b (9.1.0.441655) 64-bit (glnxa64)
                              September 7, 2016
 
  
 For online documentation, see http://www.mathworks.com/support
 For product information, visit www.mathworks.com.
  
 
 	Non-Degree Granting Education License -- for use at non-degree granting, nonprofit,
 	educational organizations only.  Not for government, commercial, or other organizational use.
 
```

but then I got the error. Do the tests run on your pc?

Ganimed, 2017-01-24:
@gabrielenava: Yes, there is still a problem with the segmentation fault. I compiled the code also in DEBUG mode, to see all the outputs and steps.
The problem is still in the destructor of the ModelState class.
The strange thing is, if you call the mex-subroutine in a normal procedure like "timeTest.m" and afterwards unload it from Matlab's memory manager with the "clear all" command, then the deallocation of the arrays in the ModelState class works flawlessly.
But in the CTest it fails at exact position. After freeing the _robotWBIModel_, the system is not able to free the following arrays of the state variable. That's really strange!
Furthermore, outside of the mexWholeBodyModel subroutine the CTest methods in Matlab causing also some sporadic segmentation faults. I suspect in this case that some random generated values are out of scope and causing these effects.
But anyway the subroutine is now more stable than before.
Let's talk about this tomorrow.

Ganimed, 2017-01-24:
P.S: My system shows mostly that the CTest has passed successfully but immediately afterwards when the subroutine will be unloaded from the system it causes a segmentation problem. More details I will explain in the report.

gabrielenava, 2017-01-26:
ctest is finally worrking! great job @Ganimed, I will close this issue as soon as we merge into master

Ganimed, 2017-01-26:
Grazie! :smile:

closed 2017-01-30 by gabrielenava