robotology-legacy/mex-wholebodymodel #74: Not used wbm_setWorldLink.m function

traversaro, 2016-08-31:
I think `wbm_setWorldLink` is unused since https://github.com/robotology/mex-wholebodymodel/blob/aa6199ff48b524d3ce7b827b6cabe144c23688ab/mex-wholebodymodel/matlab/wrappers/wbm_setWorldLink.m . 
Do you think we can remove it @Ganimed @gabrielenava ?

Ganimed, 2016-09-01:
I don't use the function *wbm_setWorkdLink()* in my project. I've avoided this function from begin on, since it was confusing me. If nobody else is using it, you can remove this function.

gabrielenava, 2016-09-01:
It is also not used in the controllers, so I think we can remove it

traversaro, 2016-09-01:
Fixed in https://github.com/robotology/mex-wholebodymodel/commit/7a0b3f130e73aff3b1e659a8059416d6c690dd4e .

closed 2016-09-01 by traversaro