robotology-legacy/mex-wholebodymodel #77: Bug: DBL_MAX in yarpWholeBodyModelV2::getJointLimitFromModel

Ganimed, 2017-01-17:
Hi guys,

yesterday I updated completely all libraries to the latest versions. After working around with my code, I have recognized that the Matlab function _wbm_jointLimits_ is returning vectors with values of +- 1.79769e+308.
These are extremely high values and does not make any sense!
Afterwards I tried to make a CTest. But the CTest fails, since in _baseLinkFrameTest.m_ the initialization of _**qjInit**_ will produce a vector of **Inf**.
```
 [jl1,jl2] = wbm_jointLimits();
 params.qjInit = jl1 + rand(25,1).*(jl2-jl1);
```
All further calculations afterwards will fail in baseLinkFrameTest.m, since the Inf-vector causes NaNs.

So I made some further research in the C++ code of the mexWholeBodyModel subroutine and I have found out that the yarpWholeBodyInterface-function _**getJointLimits()**_ returns already these high values. After a deeper look into the code I have found that the variable **DBL_MAX**  in _yarpWholeBodyModelV2::getJointLimitFromModel_ is initialized with the value 1.79769e+308. This is not correct. But I don't know where DBL_MAX will be initialized.

Can someone fix this?
Thanks.

traversaro, 2017-01-17:
@Ganimed thanks for catching this issue. 

@S-Dafarra Can you look into this when you have time? I suspect the new limits parser in iDynTree is not working, and given that you need to use it you would need to investigate it. 
The relevant code was added in https://github.com/robotology/idyntree/commit/bb0b92ed09cbc62f312d2268012d86d8875bae16 .

traversaro, 2017-01-17:
@Ganimed as a workaround, you can revert this PR https://github.com/robotology/yarp-wholebodyinterface/pull/63/files and everything should work fin again, until we fix the issue.

Ganimed, 2017-01-17:
Thanks. I will do this in the meantime.

traversaro, 2017-01-18:
@Ganimed this should be fixed by https://github.com/robotology/idyntree/issues/257 . Let us know if the problem persists.

closed 2017-01-18 by traversaro

Ganimed, 2017-01-18:
That was quick! Thanks. I will update and check it tomorrow.