robotology-legacy/mex-wholebodymodel #84: Create CI files

francesco-romano, 2017-02-27:
This is starting to become important

traversaro, 2017-02-27:
Remaining on freely available CI systems, compilation can be tested with the MCR [1] or with the Octave mex interface [2]. 
For running the tests, I think the only viable option is Octave, but probably we need to change some of the code (for example some integrators that we use are not available on Octave, and at least on octave we should use on of the available integrators). 

[1] : https://github.com/robotology/idyntree/issues/250
[2] : https://www.gnu.org/software/octave/doc/v4.0.0/Mex_002dFiles.html

traversaro, 2020-09-29:
In the end we never added CI support for this repo, but interestingly now Travis has Matlab support: https://docs.travis-ci.com/user/languages/matlab/ .

closed 2020-09-29 by traversaro