robotology-legacy/mex-wholebodymodel #85: latest mex-subroutine causes numerical problems

Ganimed, 2017-03-03:
Hi guys,

today I have found by accident a numerical problem that is causing the latest C++ code of the _mexWholeBodyModel_.

The numerical issue is rising up when the current mex-subroutine is used in combination with a _forward dynamics_ function that is calling the ```ode15s``` function of Matlab. 

The issue can be reproduced by calling the _integrateForwardDynamics.m_ (matlab/wrappers/+WBM/examples), which calls the ```fastForwardDynamics()``` function.

Before I made in the End of January all the changes and optimizations in the C++ code, the _integrateForwardDynamics.m_ worked well without printing any warnings.
Now the ```ode15s``` method prints a lot of warnings that the values are not meeting the integration tolerances like below.

```
Warning: Failure at t=8.181408e-01.  Unable to meet integration tolerances without reducing the step size below the smallest value
    allowed (1.776357e-15) at time t.
    > In ode15s (line 730)
      In integrateForwardDynamics (line 55)
```

So I guess I have something "over-optimized" in the C++ code that is causing this numerical issue. I have to compare the latest code with the previous versions in Git, to localize which change is causing this numerical problem.

_**If somebody is using the ```ode15s``` function and has numerical warnings, it is recommended for the meantime to go back to an older version, until I have found and fixed this numerical issue!**_

gabrielenava, 2017-03-06:
HI @Ganimed, when you fixed this bug in your branch I will test it on `WBM_Controller` branch to see if everything is ok

Ganimed, 2017-03-06:
On the weekend I have found and fixed the bug in _modeljacobian.cpp_. The corrected version is already uploaded on GitHub. Now the Jacobian works again correctly! You can merge the correction into the WBM-Controller branch.

But the numerical problem is still not 100-percently solved! If the _integrateForwardDynamics.m_ procedure is executed, the flood of numerical warnings is gone now. The situation is now much better, but I get still one or two warnings.

So I was checking the C++ code again. I tried to exchange some parts in the code with a different approach to see if it makes a difference. But nothing has been changed. The changes didn't made any effects. So I went back to an uploaded version state where I was absolutely sure that the Matlab-procedure was working correctly under this _"code snapshot"_. After the compilation I had again the same numerical problem with ```ode15s``` function as with the latest corrected version. So I guessed that the problem is not anymore in the C++ code of the Matlab-subroutine.
I have updated everything to the latest version and checked again. But nothing has been changed, the issue was still there. In the end I had the suspicion that _yarpWholeBodyModel.h_ is causing again the troubles. In the header-file have changed the _yarpWholeBodyModelV2_ to _yarpWholeBodyModelV1_ and recompiled everything. Afterwards the numerical warnings were gone!
So my suspicion was right.

Somewhere in a function of the _yarpWholeBodyModelV2_ interface is causing a numerical problem. In the next days I will try to find out which function in the interface is causing this issue.

**_To avoid the numerical problems with the ```ode15s``` function it is recommended in the meantime to move back to ```yarpWholeBodyModelV1```, until the bug is located and fixed in ```yarpWholeBodyModelV2```._**

traversaro, 2020-09-29:
Hi @Ganimed, if this is an issue for you feel free to open it at https://github.com/Ganimed/wbm-library, thanks!

closed 2020-09-29 by traversaro