robotology-legacy/mex-wholebodymodel #86: Clarification on the function

DanielePucci, 2017-03-20:
Hi @Ganimed @gabrielenava,
it seems to me that the function

`https://github.com/robotology/mex-wholebodymodel/blob/master/mex-wholebodymodel/matlab/utilities/%2BWBM/%2Butilities/dquat2tform.m`

computes a sort of homogeneous transformation where the associated rotation matrix is computed not from a quaternion, but from a quaternion derivative:

`https://github.com/robotology/mex-wholebodymodel/blob/master/mex-wholebodymodel/matlab/utilities/%2BWBM/%2Butilities/dquat2tform.m#L4`.

The meaning of this  operation is not completely clear to me. Can you help me understand this?

Thanks.

Ganimed, 2017-03-20:
Hi Daniele,

thanks for the advice!
The function ```dquat2tform``` is a function which does not make any sense and is wrong. This function is a relict from the very beginning and I have overseen it. You can ignore it. In the next update it will be removed from the repository.

Only the opposite direction would make a sense (orientation to quaternion rate), like ```rotm2dquat``` or ```tform2dquat```, if the _angle velocity **w**_ is known.

But a function to _"predict"_ the orientation is maybe useful and can be calculated like in the following link:
http://math.stackexchange.com/questions/189185/quaternion-differentiation

DanielePucci, 2017-03-20:
Hi @Ganimed,
thanks for the clarification. For the time being, I would avoid to put some effort in defining functions for orientation prediction.

gabrielenava, 2017-08-28:
Function removed with  https://github.com/robotology/mex-wholebodymodel/commit/8da30955676ddeb4037e997df296128bf7cd024e. Closing this issue

closed 2017-08-28 by gabrielenava