robotology-legacy/mex-wholebodymodel #90: Deprecation of codyco-superbuild

traversaro, 2018-04-18:
Hi @Ganimed, 
  how are things going on? 

As you may read in https://github.com/robotology/codyco-superbuild/issues/198, we are currently deprecating the codyco-superbuild in favor of the new robotology-superbuild, that is not including `mex-wholebodymodel`. In general, our future plan is to stop working on `mex-wholebodymodel`. What are instead your plans for `mex-wholebodymodel` ? If you still plan to work on it/maintain it, a feasible thing could be that we transfer the ownership of the repo to you.  
Let me know what you think about it, thanks. 
         Silvio

Ganimed, 2018-04-19:
Hello Silvio,

I'm fine, thank's. At the moment I'm at home but working beside for the TU-Wien and I'm still working on the mex-wholebodymodel.

No, I didn't seen the announcement. That's bad. In the meantime I have extended the WBM-Class much further for the University in Vienna. The whole-body model classes for Matlab have now much more functionalities. Currently I'm still writing on the documentations for the new library by using Sphinx.

My thoughts were, when I have finished writing the documentation and after some cleanups, to rename the "mex-wholebodymodel" repository in "WBM-Library". The new library is now really large and the new name would make more sense. The library has also now an interface to the Robotics Toolbox of Peter Corke. The interface makes it now easy to compare the performance of an algorithm with different robot models (fixed base vs. floating base).  I have also planned to add a DOI-Number to the library. That makes it easier for referencing, if the code will be used for scientific experiments.

I have also planned to contact you soon to inform you of my progress with the new library. But in this case this new information changes everything! So how can we do? In the meantime please transfer the ownership of the repo to me! Then I can proceed working on the new WBM-Library. Maybe we can integrate afterwards a fork of the mex-wholebodymodel with the new WBM-Library (and their docs) into the new robotology-superbuild, when the library is completely finished.

Please inform also your Team about that and ask them if it is possible to integrate it afterwards. The new library will be finished around summer. Will be the iDynTree-library also getting deprecated? Because I have also planned to integrate the visualizer of the iDynTree-library into the new WBM-Library.

With best regards,
Martin.

traversaro, 2018-04-19:
Hi @Ganimed , I think we can discuss more efficiently with a quick chat, I wrote you an email.

traversaro, 2018-04-19:
In the meanwhile: do not worry, nothing will change before we sorted out everything. : )

Ganimed, 2018-04-19:
Hi @traversaro, thanks for you quick response! I did not received your email. Did you have the correct address (_martin.neururer(AT)student.tuwien.ac.at_  or  _martin.neururer(AT)gmail.com_)?

Ganimed, 2019-06-03:
Hi @traversaro and @gabrielenava, I have now duplicated the complete repository to the new repository `wbm-library` at my account. Now we can merge the _yarp-interfaces_ into this new repository.

traversaro, 2020-09-29:
Thanks @Ganimed ! Then we can close this issue.

closed 2020-09-29 by traversaro