robotology-legacy/mex-wholebodymodel #91: Move this repo to Robotology Legacy?

gabrielenava, 2018-04-19:


traversaro, 2018-04-19:
Related to : https://github.com/robotology/mex-wholebodymodel/issues/90 .

gabrielenava, 2019-04-15:
@traversaro @Ganimed in the end, can we proceed in moving the repo to legacy? I remember we discussed about moving the WBM-class in another repo, am I right?

traversaro, 2019-04-15:
Ok for me.

Ganimed, 2019-04-15:
Sorry, the last time I was quite busy. Yes, let's moving my branch into a new repository. Then we can put the _mex-wholebodymodel_-repo to the legacy branch.
So, when do have time that we can merge everything together into a new repo? Thanks.

Ganimed, 2019-04-16:
Is it possible to make a new repository in _robotology_ called "wbm-library"? I don't know if I have the rights there to create a new repo. Thanks.

traversaro, 2020-09-29:
Done.

closed 2020-09-29 by traversaro