robotology-legacy/mex-wholebodymodel #92: bug in rotm2quat.m

ronb127, 2018-05-14:
Line 35:
quat(1,1) = (rotm(3,2) + rotm(2,3))*s_inv;
Should be:
quat(1,1) = (rotm(3,2) - rotm(2,3))*s_inv;

traversaro, 2018-05-14:
Thanks for reaching out @ronb127 ! 
cc @Ganimed

Ganimed, 2018-05-15:
Thank you, I will check this!

Ganimed, 2018-05-15:
I have checked the formulas. Mainly I have used the formulas from the book of [Jaz10] and in equation (3.150) at page 110 there is the same sign-bug. 

Here for example is the correct calculation of this part: http://www.euclideanspace.com/maths/geometry/rotations/conversions/matrixToQuaternion/.

Thank you for the hint I have corrected this in my branch.

[Jaz10]: Jazar, Reza N.: Theory of Applied Robotics: Kinematics, Dynamics, and Control. 2nd Edition, Springer, 2010, p. 110, eq. (3.149)-(3.152). 

@traversaro: I need the owner-rights that I can merge it to the master, thank's.

closed 2018-05-18 by Ganimed